STMicroelectronics/stm32h7xx-hal-driver #2: HAL_FDCAN_IRQHandler() : Concern with regards to value assigned to FDCAN_CCU->IR and FDCAN_CCU

rxa1031, 2020-03-17:
Dear Developer,

File: STM32h7xx_hal_fdcan.C

I am concerned about the values assigned to `variables TxEventFifoITs, RxFifo0Its, and RxFifo1Its`.

I am concerned about the values assigned to registers `FDCAN_CCU->IR` and `FDCAN_CCU->IE`.

**Observations:**

As per the STM32H743 / STM32H753 Reference manual, the `bit numbers 30 and 31 of **FDCAN_IR** is RESERVED`. Similarly the `bit numbers 30 and 31 of **FDCAN_IE** are also reserved`.

 Also the `**CCU_IE** register bits 2 through 31 are reserved`.
 

**Concerns:**

Please refer to below declaration for CAN and the shared code parts.

I want to understand why one would call `FDCAN_CCU->IE |= (((__INTERRUPT__) & CCU_IR_MASK) >> 30);` when the passed `__INTERRUPT__` will always have `unknown value for bit 30 and bit 31`?

When `HAL_FDCAN_IRQHandler()` function executes, a value gets  assigned to variables out of `TxEventFifoITs`, `RxFifo0Its`, and `RxFifo1Its`.

**My concern is that when both FDCAN_IR and FDCAN_IE have reserved bit numbers 30 and 31, then why do we want to use these bits by first copying their information to the variables and later using the values with the variables for assigning the two bits of FDCAN_CCU_IR and FDCAN_CCU_IE?**

I also observe that RxFifo1Its >> 4 is actually the value that coder will have to use inside HAL_FDCAN_RxFifo1Callback().

And that TxEventFifoITs >> 8 is actually the value that coder will have to use inside HAL_FDCAN_TxEventFifoCallback()

**MACROS:**
```
#define FDCAN_IR_MASK ((uint32_t)0x3FCFFFFFU) /*!< FDCAN interrupts mask */
#define CCU_IR_MASK   ((uint32_t)0xC0000000U) /*!< CCU interrupts mask   */

#define __HAL_FDCAN_CLEAR_FLAG(__HANDLE__, __FLAG__)             \
do{                                                              \
    ((__HANDLE__)->Instance->IR) = ((__FLAG__) & FDCAN_IR_MASK); \
    FDCAN_CCU->IR = (((__FLAG__) & CCU_IR_MASK) >> 30);          \
  }while(0)

#define __HAL_FDCAN_ENABLE_IT(__HANDLE__, __INTERRUPT__)             \
  do{                                                                \
    (__HANDLE__)->Instance->IE |= ((__INTERRUPT__) & FDCAN_IR_MASK); \
    FDCAN_CCU->IE |= (((__INTERRUPT__) & CCU_IR_MASK) >> 30);        \
  }while(0)
```

**Code part:**
I observe following code part present within HAL_FDCAN_IRQHandler() function’s body:
```
  TxEventFifoITs = hfdcan->Instance->IR & FDCAN_TX_EVENT_FIFO_MASK;
  TxEventFifoITs &= hfdcan->Instance->IE;
  RxFifo0ITs = hfdcan->Instance->IR & FDCAN_RX_FIFO0_MASK;
  RxFifo0ITs &= hfdcan->Instance->IE;
  RxFifo1ITs = hfdcan->Instance->IR & FDCAN_RX_FIFO1_MASK;
  RxFifo1ITs &= hfdcan->Instance->IE;
…
  /* Tx event FIFO interrupts management **************************************/
  if (TxEventFifoITs != 0U)
  {
    /* Clear the Tx Event FIFO flags */
    __HAL_FDCAN_CLEAR_FLAG(hfdcan, TxEventFifoITs);

    /* Tx Event FIFO Callback */
    HAL_FDCAN_TxEventFifoCallback(hfdcan, TxEventFifoITs);
  }

  /* Rx FIFO 0 interrupts management ******************************************/
  if (RxFifo0ITs != 0U)
  {
    /* Clear the Rx FIFO 0 flags */
    __HAL_FDCAN_CLEAR_FLAG(hfdcan, RxFifo0ITs);

    /* Rx FIFO 0 Callback */
    HAL_FDCAN_RxFifo0Callback(hfdcan, RxFifo0ITs);
  }

  /* Rx FIFO 1 interrupts management ******************************************/
  if (RxFifo1ITs != 0U)
  {
    /* Clear the Rx FIFO 1 flags */
    __HAL_FDCAN_CLEAR_FLAG(hfdcan, RxFifo1ITs);

    /* Rx FIFO 1 Callback */
    HAL_FDCAN_RxFifo1Callback(hfdcan, RxFifo1ITs);
  }
```
Please share your views.

Thanks,
Rajeev

rxa1031, 2020-04-16:
I had sent an email about this issue to Luca (from ST). He would have probably raised the internal request for fix.

ALABSTM, 2020-04-16:
Hi @rxa1031,

Thank you for this last comment. Indeed, the issue has been forwarded to our development teams and is currently under analysis. I will let you updated as soon as I have feedback on it.

Thank you once again for having reported this concern. Take care and stay safe.

With regards,

ALABSTM, 2020-04-16:
ST Internal Reference: 83674

rxa1031, 2020-04-19:
I was informed:
[ST] issue reported to CubeHAL team. we agree with your comments, and there are some other things that should be improved such as:
(__HANDLE __) -> Instance-> IE | = ((__INTERRUPT__) & FDCAN_IR_MASK); \
should be
(__HANDLE __) -> Instance-> IE | = (__INTERRUPT__);
as already implemented in the stm32g4xx_hal_fdcan.h file.
 
This will be fixed in next releases

Below are my observations and inputs:
My initial consideration, while raising the concern was incorrect, as I did not check that `writing 1 to a bit of FDCAN_CCU_IR and FDCAN_IR will only clear them`. `Writing 0 to any bit does not have any impact`. I am sorry for making the mistake.

I am of a different opinion compared to what the information I have recently received with regards to FDCAN_IR_MASK.

I had a quick glance at the STM32G4 Series Reference manual, April 2020, RM0440, Rev 3. I tried searching for FDCAN_CCU_IR or its alias CCU_IR, but I was unable to find it. Hence, my conclusion is that ONLY STM32G4XX code does not have to use the FDCAN_IR_MASK (as it helps break down value parts that is to be written to FDCAN_IR and FDCAN_CCU_IR).

In case of STM32H743 and STM32753:
1.	If HAL Driver continues to use one macro for writing to both the FDCAN_IX and FDCAN_CCU_IX (where X = R or E) at the same time, then I am in favor of using FDCAN_IR_MASK be it with FDCAN_CCU_IE or FDCAN_CCU_IR. The reason is that, if the mask is removed then a write to FDCAN_CCU_IR or FDCAN_CCU_IE will also impact the bits [31:30] of FDCAN_IR or FDCAN_IE (as applicable). It may be possible that one of the currently reserved bit is used in future for some other interrupt.
2.	In future reserved bits [31:30] of FDCAN_IR and FDCAN_IE may get used for some other implementation / interrupt, hence it is not advisable to continue with current implementations of __HAL_FDCAN_ENABLE_IT(),  __HAL_FDCAN_DISABLE_IT(), __HAL_FDCAN_CLEAR_IT() and __HAL_FDCAN_CLEAR_FLAG(). Either there should be separate macros for FDCAN_IE and FDCAN_CCU_IE (similarly for FDCAN_IR and FDCAN_CCU_IR) or there should be some kind of check to inform whether the value is to be updated to FDCAN_IX or FDCAN_CCU_IX (where X = R or E).

rxa1031, 2020-04-19:
I have following additional information to share:

After reviewing the inputs that I've received, I had a quick glance at the STM32G4 Series Reference manual, April 2020, RM0440, Rev 3. I tried searching for FDCAN_CCU_IR or its alias CCU_IR, but I was unable to find it. Hence, my conclusion is that ONLY STM32G4XX code does not have to use the FDCAN_IR_MASK (as it helps break down value parts that is to be written to FDCAN_IR and FDCAN_CCU_IR).

I understand writing 1 to a bit of FDCAN_CCU_IR and FDCAN_IR will only clear them. Writing 0 to any bit does not have any impact. However, this is not the case with FDCAN_CCU_IE and FDCAN_IE

Hence, I request that all four macros namely __HAL_FDCAN_ENABLE_IT(),  __HAL_FDCAN_DISABLE_IT(), __HAL_FDCAN_CLEAR_IT() and __HAL_FDCAN_CLEAR_FLAG() be modified, so as to have different macros which takes care of FDCAN_CCU_IX (where X = E or R) and FDCAN_IX (where X = E or R).

As on day the bits [31:30] of FDCAN_IX are unused, hence the coder is left shifting the FDCAN_CCU_IX bits (where X = E or R) by 30 before passing as a value to the existing macros.

What if these reserved bits [31:30] of FDCAN_IX are used for some interrupt in future, then the current implementation will become a BAD Implementation. Hence it will be good if either 8 bytes (i.e. 4 for FDCAN_IX and 4 for FDCAN_CCU_IX) or structure variable is passed to the macros so that there is not any adverse impact while making changes to the register values.

rxa1031, 2020-04-19:
Basically one should be able to update FDCAN_CCU_IE and FDCAN_IE mutually exclusively.

ALABSTM, 2020-12-14:
Hi @rxa1031,

I hope you are doing well. As for your concerns:
* As for the macro `__HAL_FDCAN_CLEAR_FLAG()` (the same applies to the macro `__HAL_FDCAN_ENABLE_IT`), it has been defined to handle **both** registers `FDCAN_IR` and `FDCANCCU_IR` using **one signle** parameter `__FLAG__`:
  * As you pointed out, register `FDCAN_IR` has **2** reserved bits (30 and 31), while register `FDCANCCU_IR` has only **2** used bits (0 and 1).
  * The idea is that parameter `__FLAG__` is made up of two parts:
    * bits 0 to 29 holding data to be written in register `FDCAN_IR`.
    * bits 30 and 31 holding data to be written in register `FDCANCCU_IR`.
  * The two parts of the `__FLAG__` parameters are extracted thanks to the masks  `FDCAN_IR_MASK` and `CCU_IR_MASK` as shown in the code snippets below. Notice also the **shift** to the right before the storage in the `FDCANCCU_IR` register.
  * I agree with you that if one day bits 30 and 31 of register are to be used, the code implementation has to be changed. However, this can be made in a way **totally transparent** to the user and with **preserving** backward compatibility.

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Inc/stm32h7xx_hal_fdcan.h#L1528-L1529
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Inc/stm32h7xx_hal_fdcan.h#L1855-L1859

* As for the variables `TxEventFifoITs`, `RxFifo0Its`, and `RxFifo1Its`, and their use within the callbacks, this is up to the user to decide of the way they see it best appropriate to use them. This includes shifting their content by 4 or by 8.
Regarding the shifting itself, I am tempted to think it can be avoided:
  * Indeed, within each one of the above mentioned variables, bit positions are preserved thanks to the masks used, as shown in the code snippets below.
  * These masks being defined using the **CMSIS bits** definitions, we could think about using them directly as suggested at the end of this comment.

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_fdcan.c#L210
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_fdcan.c#L5273-L5274

```c
void HAL_FDCAN_RxFifo1Callback(FDCAN_HandleTypeDef *hfdcan, uint32_t RxFifo1ITs)
{

  ...

  if ((RxFifo1ITs & FDCAN_IR_RF0L) != 0U)
  {
    /* Treatment in case of FDCAN_IR_RF0L interrupt */
  }

  ...
}
```

I hope this answers your questions. Thank you again for having shared your concerns. Please allow me to close this issue now.

With regards,

closed 2020-12-14 by ALABSTM