STMicroelectronics/stm32h7xx-hal-driver #3: HAL_FDCAN_GetRxMessage() evaluates GetIndex value for Rx FIFO overwrite scenario.

rxa1031, 2020-04-01:
Dear HAL Driver developer,

STM32H753 Reference Manual informs below:

When an Rx FIFO is operated in overwrite mode and an Rx FIFO full condition is signaled, reading of the Rx FIFO elements should start at least at get index + 1. The reason for that is that it can happen that a received message is written to the message RAM (put index) while the CPU is reading from the message RAM (get index). In this case inconsistent data may be read from the respective Rx FIFO element. Adding an offset to the get index when reading from the Rx FIFO avoids this problem.

However the below code which evaluates the GetIndex and then the Read RxAddress does not seem to test for Overrun condition. As per above information I understand that code need to be modified to first check for overrun condition is needed. Then in case of overrun condition the GetIndex should be incremented by 1 (based on rollover condition) and then used for calculating RxAddress:

```      /* Check that the Rx FIFO 0 is not empty */
      if ((hfdcan->Instance->RXF0S & FDCAN_RXF0S_F0FL) == 0U)
      {
        /* Update error code */
        hfdcan->ErrorCode |= HAL_FDCAN_ERROR_FIFO_EMPTY;

        return HAL_ERROR;
      }
      else
      {
        /* Calculate Rx FIFO 0 element address */
        GetIndex = ((hfdcan->Instance->RXF0S & FDCAN_RXF0S_F0GI) >> FDCAN_RXF0S_F0GI_Pos);
        RxAddress = (uint32_t *)(hfdcan->msgRam.RxFIFO0SA + (GetIndex * hfdcan->Init.RxFifo0ElmtSize * 4U));
      }```

Please share your views.

Regards,
Rajeev

ALABSTM, 2020-04-16:
Hi @rxa1031,

Thank you for this report. It has been forwarded to our development teams. I will be back to you as soon as they provide me with their feedback. Take care and stay safe.

With regards,

ALABSTM, 2020-04-16:
Hi @rxa1031,

Our development teams confirmed the issue you reported. Actually, it has been already identified and fixed. This fix will be available soon hopefully. We cannot share a date for the moment. Thank you for your comprehension.

With regards,

ALABSTM, 2020-04-16:
ST Internal Reference: 83793

rxa1031, 2020-04-16:
I had sent an email about this issue to Luca (from ST). He would have probably raised the internal request for fix.

Thanks for sharing the information.

closed 2021-04-14 by ALABSTM