STMicroelectronics/stm32h7xx-hal-driver #4: Incorrect value assigned to ProtocolStatus->Activity inside HAL_FDCAN_GetProtocolStatus() in file STM32Cube_FW_H7_V1.7.0\Drivers\STM32H7xx_HAL_Driver\Src\stm32h7xx_hal_fdcan.c

rxa1031, 2020-04-13:
Dear Developer,

Incorrect value is assigned to ProtocolStatus->Activity inside HAL_FDCAN_GetProtocolStatus() in file STM32Cube_FW_H7_V1.7.0\Drivers\STM32H7xx_HAL_Driver\Src\stm32h7xx_hal_fdcan.c

Please change:

 ProtocolStatus->Activity = (StatusReg & FDCAN_PSR_ACT);

To:

 ProtocolStatus->Activity = ((StatusReg & FDCAN_PSR_ACT) >> FDCAN_PSR_ACT_Pos);

Regards,
Rajeev

rxa1031, 2020-04-16:
I had sent an email about this issue to Luca (from ST). He would have probably raised the internal request for fix.

ALABSTM, 2020-04-29:
Hi @rxa1031,

Thank you for this other report. I confirm this issue has been logged into our internal bug tracking system as this was the case of the other two.

Your reports will always be the welcome. May I just ask you to avoid sending the same report through more than one support channel? Thank you for your comprehension.

Take care and stay safe.

With regards,

ALABSTM, 2020-04-29:
ST Internal Reference: 84743

ALABSTM, 2020-04-29:
Hi @rxa1031,

Our development teams checked your report and indicated **there is no issue**. Here are the details of their answer:

"Activity is a uint32_t in the structure FDCAN_ProtocolStatusTypeDef. It can be one of the following values:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/b1c3ba014904de02274c7ff901f0226ad6f20238/Inc/stm32h7xx_hal_fdcan.h#L1192-L1195
As you may notice those values are shifted by default. So there is no need to shift them again."

With regards,

closed 2020-07-20 by ALABSTM