STMicroelectronics/stm32h7xx-hal-driver #5: HAL UART Driver issue wherein the last byte is received incorrectly depending on the data of the last byte

rxa1031, 2020-04-24:
Hi @ASELSTM , @ALABSTM ,

HAL Driver (internal) version 1.8.0. Using Latest STM32CubeMX version 5.6.1 with STM32H7 latest package (1.7.0).

It is observed that even after adding fix informed under https://github.com/STMicroelectronics/stm32h7xx_hal_driver/issues/1 , the **captured last byte is incorrect when the baud-rate is increased to 921600 bps.**

UART1 input clock is set to 180 MHz on STM32H743XI Revision Y. Other settings are shown below:

[image]

It is observed that when bytes \x55\xA**A** are sent to microcontroller (by Laptop), they are received as \x55\xA**8**.
Similarly the 2 byte data \x55\xA**9** is received as \x55\xA**1**

The data \x55\xFF is however received correctly.

_The above issue is not observed when using baudrate of 115200 bps._

Regards.
Rajeev

rxa1031, 2020-04-24:
I forgot to mention that I am using function HAL_UART_Receive_IT() for receiving two bytes of data.

ASELSTM, 2020-04-29:
Hi @rxa1031,

Thank you for this one other issue you reported

Would you please share the whole CubeMx project you have used to reproduce the issue.

With regards,

rxa1031, 2020-04-29:
Hi,

Attaching ZIP which contains both the project and IOC file. [UART-IT-with-921600-bps-send-for-review-20200429-2013.zip](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/files/4552422/UART-IT-with-921600-bps-send-for-review-20200429-2013.zip)

The IOC file helps set up USART1 for Asynchronous communication with 115200 8-N-1

The code generated via the IOC file is updated to:
1. Use Baudrate of 921600 bps instead of 115200 bps
2. To fix the #1 bug, below code is added inside function MX_USART1_UART_Init() in USER CODE USART1_Init 2 area.
			/* Process Locked */
			__LOCAL_HAL_LOCK(huart);
			if (huart->FifoMode == UART_FIFOMODE_ENABLE)
			{
				huart->NbRxDataToProcess *= (2U);
			}
			/* Process Unlocked */
			__LOCAL_HAL_UNLOCK(huart);
 
Regards,
Rajeev

rxa1031, 2020-05-15:
Hi @ASELSTM 

Were you able to reproduce the issue?

Regards,
Rajeev

ASELSTM, 2020-06-19:
Hi @rxa1031 ,

Sorry for the delay.

We have been trying to reproduce the issue.  Unfortunately, we did not obtain the behavior described above. The project is working well and the data sent from the laptop are correctly received by the microcontroller.

With regards,

rxa1031, 2020-06-19:
Hi @ASELSTM  ,

Did you use the ZIPPED code that I shared and the STM32H743I-EVAL board?

If not, then I hope you have used the exact STM32CubeMX settings that I have shared and the released version of STM32CubeMX.

I hope you have tested the issue using the Interrupt mode, FIFO enabled and RX_FIFO_DEPTH to 16U (and not 8U).

Please share the code that you used for testing. I shall give it a try.

Thanks,
RXA1031

rxa1031, 2020-06-19:
Hi @ASELSTM ,

If possible please also share the IOC file.

Regards,
@rxa1031

ASELSTM, 2020-06-22:
Hi @rxa1031,

Yes, we have used the zipped code that you have shared with us and tested on a STM32H743I-EVAL board.

With regards,

ASELSTM, 2020-07-23:
Hi @rxa1031,

Please allow me to close this thread as no activity has been registered since a month. Thank you for your comprehension. 

With regards,

closed 2020-07-23 by ASELSTM