STMicroelectronics/stm32h7xx-hal-driver #7: Bug in stm32h7xx_hal_rcc_ex.c - wrong #define for HRTIM1 (version 1.8)

pavel-a, 2020-06-30:
In file stm32h7xx_hal_rcc_ex.c  function HAL_RCCEx_PeriphCLKConfig

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/b1c3ba014904de02274c7ff901f0226ad6f20238/Src/stm32h7xx_hal_rcc_ex.c#L1474

Instead of #if defined(**HRTIM**) should be #if defined(**HRTIM1**) 

Result: HRTIM1 is configured incorrectly.

[Reported on the STM32 forum](https://community.st.com/s/question/0D53W0000080rkUSAQ/problem-with-hrtim-nucleoh743zi2frequency-of-timer-should-be-480-mhz-but-it-is-still-240mhz)  by [AndiAndi
](https://community.st.com/s/profile/0050X000007SCuxQAG)

ALABSTM, 2020-07-01:
Duplicate of issue [STM32CubeH7/#46](https://github.com/STMicroelectronics/STM32CubeH7/issues/46).

ALABSTM, 2020-07-01:
Hi @pavel-a,

Thank you once again for all your reports. This point has been transmitted to our development teams. I will get to you once I have their answer.

Kind regards,

closed 2020-07-04 by pavel-a