STMicroelectronics/stm32h7xx-hal-driver #8: Ethernet: Add ability to explicitly free specific descriptors or release descriptor with new address

vielster, 2020-09-17:
In embedded IP stack usage (e.g. lwip), which is typically implemented as zero-copy, the presently available interfaces for recycling RxDescriptors is problematic. Ideally, when a packet is received and read out of the HAL, the driver code would pass this buffer to the stack, and would only release the descriptor back to the HAL driver when the stack releases this buffer (e.g. through a `pbuf_free()` call).  Between pushing this message into the stack, and the stack _freeing_ this buffer back to the driver (and thus the descriptor to the HAL), it's possible that additional packets are received that go through this same process.

Given the scenario above, in a multi-threaded environment, there is no way to guarantee the order the packets are _freed_ is the same order in which they are _received_. The only support approach in the HAL driver is to free all consumed RX descriptors through the `HAL_ETH_BuildRxDescriptors()` call, but as the freeing may happen in a different order, this approach may cause descriptors to get re-used before the data has been consumed.

Ideally, the user would be able to release RX descriptors in any order and as singalled by the stack through `pbuf_free()` like calls.  A few possible ideas for this would be...

1. Simplest API would be to release the last descriptor (single), but with a new buffer address, for example `HAL_ETH_ReleaseOneRxDescriptor(ETH_HandleTypeDef *heth, void *new_buffer_address)`. This could optionally be extended to take an array or even something like an `ETH_BufferTypeDef` linked list, `HAL_ETH_ReleaseRxDescriptors(ETH_HandleTypeDef *heth, ETH_BufferTypeDef *buffers) although implementing this from the user code with multiple calls to `ReleaseOne` might actually be a little more straightforward than creating the linked list.

1. API such as `HAL_ETH_ReleaseRxDescriptor(ETH_HandleTypeDef *heth, uint32_t index)`.  For this API to work, the user would need to be able to also get the descriptor indices associated with the  for `HAL_ETH_GetRxDataBuffer()` possibly it could be embedded in the `ETH_BufferTypeDef` or through a separate `HAL_ETH_GetRxDescriptorInfo()` such that it could then be tracked and freed when available.

ALABSTM, 2021-02-08:
Hi @vielster,

Thank you for the point you raised and the proposals. Please allow me to introduce @CHAMSTM who will look deeper into the subject.

With regards,

pavel-a, 2021-04-26:
After looking at [FreeRTOS+ fork of the ETH driver]( https://github.com/FreeRTOS/FreeRTOS-Plus-TCP/tree/main/portable/NetworkInterface/STM32Hxx) I think that most of descriptors & buffers manipulation should be moved out  of the "HAL" ETH driver to "middlewares" dependent file (separate for LwIP, FreeRTOS etc.).

ALABSTM, 2021-12-08:
Hi @vielster,

Thank you for this detailed report. We just made available a fully reworked ETH HAL driver via pull-request #16. You can use this new driver for your applications. Please note that this new ETH HAL driver **breaks the compatibility** with the previous one.

Regarding the point you raised, it should be no more applicable as the driver's logic has changed almost totally.  Please feel free to provide your feedback [here](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/pull/16) once you have performed your tests. 

With regards,

closed 2022-01-20 by ALABSTM