STMicroelectronics/stm32h7xx-hal-driver #9: EXTI reset values for IMR1-3 seems to be wrong

Viatorus, 2020-10-27:
**Describe the bug**
In 

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_ll_exti.c#L115-L117

the reset values are different then described inside the reference manual, _20.6.27 EXTI register map_, table 126.

**Screenshots**
[image]

ASELSTM, 2020-12-14:
Hi @Viatorus,

Thank you for this report. The issue has been forwarded to our development teams. We will get back to you as soon as they provide us with their feedback.

With regards,

ASELSTM, 2020-12-15:
Hi @Viatorus,

According to our development team, LL_EXTI_DeInit() is rather performing a functional rest of the EXTI lines to disable all EXTI lines interrupts and events. That's why the code is setting all the EXTI registers to zero instead of setting them to their default reset values.
Actually, for the STM32H7 case, the hardware reset values of IMRx and EMRx registers are enabling the EXTI lines which is not the purpose of the LL_EXTI_DeInit() function.

On the other hand, the development team is confirming that the comments related to IMRx and EMRx registers reset are a quite confusing. Therefore, comments will be updated as following to avoid any ambiguity : 

```diff
-  /* Interrupt mask register set to default reset values */
+ /* Interrupt mask register reset */
  LL_EXTI_WriteReg(IMR1, 0x00000000U);
  LL_EXTI_WriteReg(IMR2, 0x00000000U);
  LL_EXTI_WriteReg(IMR3, 0x00000000U);

-  /*  Event mask register set to default reset values */
+ /*  Event mask register reset */
  LL_EXTI_WriteReg(EMR1, 0x00000000U);
  LL_EXTI_WriteReg(EMR2, 0x00000000U);
  LL_EXTI_WriteReg(EMR3, 0x00000000U);
```

The fix will be implemented and made available in the frame of  a future release. Thank you once again for your contribution.

With regards,

ASELSTM, 2020-12-15:
ST Internal Reference: 98251

Viatorus, 2021-01-04:
So on SW reset into bootloader/into application, the values should set to zero.

But are they set automatically to "hardware default' like described in table 126? Otherwise there would be a difference between a HW reset (power off/on) and a SW reset.

ASELSTM, 2021-01-07:
Hi @Viatorus,

Actually, IMRx and EMRx register are set to their default reset values described in the table 126 when performing a HW reset or a SW reset. It is only at the level of the LL_EXTI_DeInit() function that IMRx and EMRx register are set to zero in order to disable all EXTI lines interrupts and events. 

With regards,

ASELSTM, 2022-01-12:
Hi @Viatorus,

The issue you reported has been fixed in the frame of version v1.10.0 of the STM32CubeH7 published on GitHub.

Please allow me then to close this thread and thank you again for having reported.

With regards,

closed 2022-01-12 by ASELSTM