STMicroelectronics/stm32h7xx-hal-driver #10: Issues in LL_RTC_TIME_Init() and LL_RTC_DATE_Init()

mft-mtc, 2020-12-10:
Both routines call LL_RTC_DisableInitMode(RTC); instead of LL_RTC_DisableInitMode(RTCx);

In LL_RTC_DATE_Init() the asserts assume binary values, but IS_LL_RTC_MONTH() compares the value to #define values that are coded as BCD.

I suggest recoding both IS_LL_RTC_MONTH() and IS_LL_RTC_WEEKDAY() to compare to binary ranges rather than comparing to the individual #define values.

ALABSTM, 2020-12-15:
Hi @mft-mtc,

Thank you for your report. You are absolutely right about both points. I will forward them to our development teams. Fixes will be published in the frame of a future release.

With regards,

ALABSTM, 2020-12-15:
ST Internal Reference: 98286

ALABSTM, 2020-12-16:
Hi @mft-mtc,

Regarding the second point you reported, I would draw your attention to the following:
* It is true the months are defined as BCD values, as you pointed out.
* However, **before** passing the "month" to the `assert_param()` function, it is converted to binary as shown in the code snippet below.

This explains why the `assert_param()` would correctly do its job whether the month's value is binary- or BCD-coded.

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_ll_rtc.c#L398-L401

With regards,

ALABSTM, 2020-12-16:
Related to [STM32CubeG0/#4](https://github.com/STMicroelectronics/STM32CubeG0/issues/4).

mft-mtc, 2020-12-16:
Line 400 will never be executed. If the input data is in binary, the month values will range from 0x1 to 0xc, so the second compare will always be false.

Lines 405 and 411 are the problematic uses.

On another note, when using Keil's version 6 (CLang based) compiler and enabling all warnings, the driver code generates quite a lot of warnings. Is there an effort to clean up the code to remove these warnings?

ALABSTM, 2021-01-11:
Hi @mft-mtc,

The thing might look quite tricky. I hope this clarifies the point. Actually, line [400](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_ll_rtc.c#L400) is intended to be executed **only** in case the two conditions below are true:
* `RTC_Format` is set to `LL_RTC_FORMAT_BIN`
* `RTC_DateStruct->Month` is set to one of the predefined **BCD** values below:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Inc/stm32h7xx_ll_rtc.h#L328-L330

Regarding the warning messages, support of Keil's ARMCC6 compiler is ongoing and should be addressed in a future release. We cannot share a date for the moment, but the warning messages you pointed out should disappear.

With regards,

ALABSTM, 2021-12-15:
Hi @mft-mtc,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.10.1** of the stm32h7xx_hal_driver published recently on GitHub. Thank you again for having reported.

With regards,

closed 2021-12-15 by ALABSTM