STMicroelectronics/stm32h7xx-hal-driver #11: Redefinition of HAL_EnableDBGSleepMode in HAL and legacy HAL

drdott, 2020-12-22:
**Describe the set-up**
 * NUCLEO-745ZI-Q
 * STM32CubeMX 6.1.0
 * STM32Cube MCU Package for STM32H7 Series v1.8.0
 * IAR 8.50
 * Ceedling 0.30

There is a redefinition of some function names between the HAL drivers and the legacy definitions.

The real function prototypes are in stm32h7xx_hal.h, lines 1082-1087:
void HAL_EnableDBGSleepMode(void);
void HAL_DisableDBGSleepMode(void);
void HAL_EnableDBGStopMode(void);
void HAL_DisableDBGStopMode(void);
void HAL_EnableDBGStandbyMode(void);
void HAL_DisableDBGStandbyMode(void);

And the code exists for those functions in stm32h7xx_hal.c, lines 917-967.

However, the legacy include legacy/stm32_hal_legacy.h redefines them, lines 1492-1497:
#define HAL_EnableDBGSleepMode HAL_DBGMCU_EnableDBGSleepMode
#define HAL_DisableDBGSleepMode HAL_DBGMCU_DisableDBGSleepMode
#define HAL_EnableDBGStopMode HAL_DBGMCU_EnableDBGStopMode
#define HAL_DisableDBGStopMode HAL_DBGMCU_DisableDBGStopMode
#define HAL_EnableDBGStandbyMode HAL_DBGMCU_EnableDBGStandbyMode
#define HAL_DisableDBGStandbyMode HAL_DBGMCU_DisableDBGStandbyMode

The legacy include legacy/stm32_hal_legacy.h is always brought into the build by stm32h7xx_hal_def.h.
stm32h7xx_hal_def.h is #included  by some of the modules (e.g. stm32h7xx_hal_rcc.h).

Those "HAL_DBGMCU_*" definitions only exist in the HAL on those lines in legacy/stm32_hal_legacy.h.

In my case this causes issues with automated unit test mock function generation using Ceedling, as it generates a mock function called HAL_DBGMCU_EnableDBGSleepMode() instead of the expected HAL_EnableDBGSleepMode().

The file stm32h7xx_hal_def.h should probably not #include legacy/stm32_hal_legacy.h.

ALABSTM, 2021-08-02:
Hello @drdott,

Thank you for this report. The issue you pointed out is confirmed. A fix will be published in a future release. Thank you again and excuse this somewhat late reply.

With regards,

ALABSTM, 2021-08-02:
ST Internal Reference: 111423

ALABSTM, 2021-12-20:
Hi @drdott,

I hope you are fine. The issue you reported has been fixed in the frame of version **v1.9.1** of the STM32CubeH7 published recently on GitHub, as you can see below. Actually, both `stm32h7xx_hal.c` and `stm32h7xx_hal.h` files were using legacy functions' names. Thank you again for having reported.

With regards, 

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Inc/stm32h7xx_hal.h#L1081-L1086

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L917

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L926

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L936

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L945

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L954

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/9439ae79fd1ead4a3fe75016912e29e8acabf4da/Src/stm32h7xx_hal.c#L963

closed 2021-12-20 by ALABSTM