STMicroelectronics/stm32h7xx-hal-driver #12: Memory alignment issues in HAL_SPI_TransmitReceive

zbychl, 2021-02-08:
I've encountered problem with HAL_SPI_TransmitReceive() function.
Issue is about wrong memory alignment. It is important when receiving data in slave mode especially.
I utilize STM32 SPI in slave mode in my project. SPI configuration is as followed:
    spi_handle.Instance = (SPI_TypeDef*)dev_config->base;
    spi_handle.Init.Mode = SPI_MODE_SLAVE;
    spi_handle.Init.Direction = SPI_DIRECTION_2LINES;
    spi_handle.Init.DataSize = **SPI_DATASIZE_8BIT**;
    spi_handle.Init.CLKPolarity = SPI_POLARITY_LOW;
    spi_handle.Init.CLKPhase = SPI_PHASE_2EDGE;
    spi_handle.Init.NSS = SPI_NSS_HARD_INPUT;
    spi_handle.Init.FirstBit = SPI_FIRSTBIT_MSB;
    spi_handle.Init.TIMode = SPI_TIMODE_DISABLE;
    spi_handle.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
    spi_handle.Init.CRCPolynomial = 7;
    spi_handle.Init.CRCLength = SPI_CRC_LENGTH_DATASIZE;
    spi_handle.Init.NSSPMode = SPI_NSS_PULSE_DISABLE;

First of all HAL_SPI_TransmitReceive() function API doesn't put any restriction on TX or RX buffer alignment. Both are pointer to uint8_t. Which means user can provide any buffer here.

My issue is about these pieces of code:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1446
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1456

I got issue reproduction when master on SPI sends data pretty fast (SCK=8MHz). Then even I configured SPI for 8bit mode it is possible to fall into case where there are two bytes in FIFO: https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1451
And when hspi->pRxBuffPtr is not aligned to 16bits I got HardFault (BusFault, PRECISERR).

I think it would be safer to copy data with memcpy() to user buffer.

ASELSTM, 2021-02-22:
Hi @zbychl,

Thank you for this report. Unfortunately, we couldn't reproduce the described issue. If the issue is systematically reproducible on your side, would you please give us more details and share the project so that we can reproduce the same behavior.

Would you also give us more details about the compiler you are using and the IDE you are working with.

With regards,

zbychl, 2021-02-26:
Hi,
Thanks a lot for response. 
I am using your STM32CubeIDE.
I am afraid I can't share my project. But going to reproduce this issues based on simple project generated by your IDE.
Thanks!

ASELSTM, 2021-06-29:
Hi @zbychl,

Any updates about how to reproduce this issue to move this discussion forward?

Best regards,

zbychl, 2021-09-17:
Hi,

Sorry for late response.
Will provide more details soon.

Thanks!

ASELSTM, 2021-11-30:
Hi @zbychl,

Please allow me to close this thread. You can reopen it at any time if you have details to provide us to help you.
Thank you for your comprehension and thank you once more for your contribution.

With regards,

closed 2021-11-30 by ASELSTM