STMicroelectronics/stm32h7xx-hal-driver #13: Overflow and underflow issue in HAL_SPI_TransmitReceive()

zbychl, 2021-02-08:
I've encountered issue in HAL_SPI_TransmitReceive() about underflow (hspi->RxXferCount) and use buffer overflow (hspi->pRxBuffPtr).  I got the reproduction for SPI in master mode.
There are SPI configuration details:
    spi_handle.Instance = (SPI_TypeDef*)dev_config->base;
    spi_handle.Init.Mode = SPI_MODE_SLAVE;
    spi_handle.Init.Direction = SPI_DIRECTION_2LINES;
    spi_handle.Init.DataSize = SPI_DATASIZE_8BIT;
    spi_handle.Init.CLKPolarity = SPI_POLARITY_LOW;
    spi_handle.Init.CLKPhase = SPI_PHASE_2EDGE;
    spi_handle.Init.NSS = SPI_NSS_HARD_INPUT;
    spi_handle.Init.FirstBit = SPI_FIRSTBIT_MSB;
    spi_handle.Init.TIMode = SPI_TIMODE_DISABLE;
    spi_handle.Init.CRCCalculation = SPI_CRCCALCULATION_DISABLE;
    spi_handle.Init.CRCPolynomial = 7;
    spi_handle.Init.CRCLength = SPI_CRC_LENGTH_DATASIZE;
    spi_handle.Init.NSSPMode = SPI_NSS_PULSE_DISABLE;

In my case SPI master transmits data with long data blocks (~16bytes) and pretty fast (SCK=8MHz). I am interested with first three bytes only. So just run HAL_SPI_TransmitReceive() with length=3. And from time to time I got two bytes in FIFO. Sometimes it results with hspi->RxXferCount underflow here:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1459

When it happens loop here makes much more data is received and copied to user buffer.
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1409

All of that results with HardFault (BusFault, IMPRECISERR) in my case.

I think there shall be check about underflow in here:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1418
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/d8461b980b59b1625207d8c4f2ce0a9c2a7a3b04/Src/stm32h7xx_hal_spi.c#L1429

KORKAD, 2021-12-10:
Hello @zbychl,
I think you have synchronization issue between Master and Slave... the master is so faster than the slave.
Try to reduce the rcc clock frequency from master side to give more time for the slave to prepare his data.
With Regards
KORKAD

ASELSTM, 2022-02-09:
Hi @zbychl,

Please allow me to close this thread as there is no activity since a couple of months. You can reopen it at any time if you have more details so that we can help you.

With regards,

closed 2022-02-09 by ASELSTM