STMicroelectronics/stm32h7xx-hal-driver #19: Ethernet HAL stopping transmitting

ashchepkov, 2022-03-10:
Hello!
Function HAL_ETH_Transmit_IT() from file stm32h7xx_hal_eth.c first check availability of current transmit buffer, then prepare descriptor and increment Tail pointer without checking who own next buffer. In case of transmitting many data this behavior may cause Head == Tail and DMA will stopped. When HAL_ETH_Transmit_IT() will be called next time, current transmit buffer will be owned be DMA and function return error without transmitting.

ASELSTM, 2022-04-08:
Hi @ashchepkov,

Could you please confirm that you are using the recently reworked ETH HAL driver proposed within this pull-request [stm32h7xx_hal_driver#16](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/pull/16) ?

With regards,

ashchepkov, 2022-04-09:
Hello!
I used in my project not last version of drivers, but i don't see difference in HAL_Transmitt_It()

ASELSTM, 2023-05-02:
Hi @ashchepkov,

Please allow me to close this thread as no activity. You may reopen it at anytime if you have any details to share with us in order to help you to solve the issue. Thank you for your comprehension.

With regards,

closed 2023-05-02 by ASELSTM