STMicroelectronics/stm32h7xx-hal-driver #21: No SPDX license identifiers

sebhub, 2022-03-11:
This software is intended to be used by applications using STMicroelectronics chips. It may be integrated in project-specific repositories. In order to manage open source software licenses, the software industry established the [Software Data Package Exchange](https://spdx.dev) standard which is an international open standard (ISO/IEC 5962:2021). It would be really nice if STMicroelectronics would support this and add SPDX license identifiers to the source files.

ALABSTM, 2022-04-05:
Hi @sebhub,

Thank you for this interesting proposal. It has been forwarded to our quality teams. I will get back to you as soon as I have their feedback.

With regards,

sebhub, 2022-04-05:
Thanks, this is related to https://github.com/STMicroelectronics/STM32CubeH7/issues/205. It would be nice if the code provided by STMicroelectronics could stay under the BSD-3-Clause license (instead of Apache License 2.0).

ALABSTM, 2022-04-13:
Hi @sebhub,

Our quality team found the proposal interesting. It may eventually be considered for future releases.

Your request has been logged internally. I will get back to you should there be any update.

With regards,

ALABSTM, 2022-04-13:
ST Internal Reference: 126395

ALABSTM, 2023-04-18:
Hi @sebhub,

I hope you are doing well. Back to you about your request. There is no plan in the near future to support SPDX identifiers, maybe in the long term. Anyway, the point is logged in our internal database.

I will get back to you should there be any progress. In the meanwhile, please allow me to close this issue. Thank you for your comprehension.

With regards,

closed 2023-04-18 by ALABSTM