STMicroelectronics/stm32h7xx-hal-driver #26: Function 'SPI_WaitOnFlagUntilTimeout': argument order different between definition and declaration

escherstair, 2022-04-15:
For the function `SPI_WaitOnFlagUntilTimeout`, the argument order is different between declaration and definition (`Timeout` and `Tickstart` swapped)

`SPI_WaitOnFlagUntilTimeout` declaration
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/8e7eb8dba8bee5e4c260c6fc724437502e98fc76/Src/stm32h7xx_hal_spi.c#L170-L171

`SPI_WaitOnFlagUntilTimeout` definition
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/8e7eb8dba8bee5e4c260c6fc724437502e98fc76/Src/stm32h7xx_hal_spi.c#L3827-L3828

ASELSTM, 2022-04-27:
ST Internal Reference: 127122

ASELSTM, 2023-04-06:
Hi @escherstair,

Thank you for your contribution. This issue has been fixed in the frame of version v1.11.1 of the stm32h7xx_hal_driver. Please allow me then to close this thread.

With regards,

closed 2023-04-06 by ASELSTM