STMicroelectronics/stm32h7xx-hal-driver #29: HAL_UART_RegisterRxEventCallback fails if there is a transmit ongoing

dergentk, 2022-08-30:
**Description**
It is impossible to set an RxEventCallback (with `HAL_UART_RegisterRxEventCallback`) if a transmit is active on the port.

**How To Reproduce**
start a UART transmit with `HAL_UART_Transmit_IT`. 
then immediately thereafter call `HAL_UART_RegisterRxEventCallback`. That call _will_ fail.

**Analysis**
This is caused by the fact that `HAL_UART_RegisterRxEventCallback` checks the value of the `gState` field instead of the `RxState` like all other receive releated functions do. I don't see a reason for this difference. As far is I can tell the RxEventCallback is unrelated to transmit so checking `gState` appears to be a mistake.

**My Solution**
Changing `gState` in `HAL_UART_RegisterRxEventCallback` and `HAL_UART_UnregisterRxEventCallback` to `RxState` makes the problem go away, as expected. And it did not cause unwanted side effects in my testing.

ASELSTM, 2023-07-27:
ST Internal Reference: 158193

ASELSTM, 2023-07-27:
Hi @dergentk,

First excuse this late answer. Indeed, you are right, the check should be done on `RxState` and not on `gState`. This will be fixed and integrated in a future release. Thank you for reporting this and thank you fir your contribution.

With regards,

TOUNSTM, 2024-01-25:
Fixed in https://github.com/STMicroelectronics/stm32h7xx_hal_driver/commit/a52bb3a5155f180d94d22ba7ebf26e9de2456b61

closed 2024-01-25 by TOUNSTM