STMicroelectronics/stm32h7xx-hal-driver #30: SDMMC stuck with HAL_SD_ERROR_DATA_TIMEOUT

daniel-starke, 2022-10-04:
**Describe the set-up**
SD card socket connected without external pull-ups (only internal ones) to a NUCLEO-H723ZG board. This made the data lines instable. Adding an external logic analyzer for example brought the lines to a working level. Switching from the non-working variant after `HAL_SD_ERROR_DATA_TIMEOUT` online to the a working variant failed while being stuck with `HAL_SD_ERROR_DATA_TIMEOUT`.

**Describe the bug**
I have observed cases like above where `HAL_SD_ERROR_DATA_TIMEOUT` was set via `SDMMC_FLAG_DTIMEOUT` and `HAL_SD_Abort()` + `HAL_SD_Init()` could not recover from this.

**How To Reproduce**
1. Provoke a `HAL_SD_ERROR_DATA_TIMEOUT` error.

2. Correct the hardware setup online (i.e. by adding pull-ups without a system reset).

3. Use `HAL_SD_Abort()`, `HAL_SD_DeInit()` and `HAL_SD_Init()` to re-initialize the device.

4. Initialization fails and the error remains.

**Additional context**
I suggest to call `SDMMC_CmdStopTransfer()` unconditionally within `HAL_SD_Abort()` to allow proper error recovery. This issue may be also present in other STM32 series HAL like STM32F7.

KRASTM, 2024-06-05:
Hello @daniel-starke,

Sorry for the very late response, I hope you are well.
I want to know if you still got this problem even after the update of the firmware during this time.

With regards.

daniel-starke, 2024-06-05:
I do not have the test setup anymore so it is hard for me to try out at the point. Would you like me to do a code review instead?

KRASTM, 2024-06-05:
Hello @daniel-starke,

Since you do not have the setup anymore, I would like to share some information related to the issue.

Usually, it's recommended to use external Pull up to get a better and stable operation, already other board equipped by SD Card contain Pull ups on the lines.
I shared your proposal with our development team, they don't accept it because we use `HAL_SD_Abort()` only when something is ongoing, and the state is busy. So, the call of **SDMMC_CmdStopTransfer()** must be conditionally within **HAL_SD_Abort()**.

Finally, please allow me to close this issue, thank you for your comprehension.

With regards.

daniel-starke, 2024-06-05:
Hello @KRASTM,

The issue is that in case of a `HAL_SD_ERROR_DATA_TIMEOUT` error (the missing pull-up was just an easy example on how to reproduce it) the HAL layer goes out of sync with the PHY. HAL falsely assumes that the PHY has finished. As is has not and HAL performs no appropriate reset the PHY remains in the error state until system reset. Feel free to use another mechanism to recover from that state. Calling `SDMMC_CmdStopTransfer()` unconditionally within `HAL_SD_Abort()` was an easy workaround which worked for me.

KRASTM, 2024-06-06:
Hello @daniel-starke,

Thank you for the explanation and the report, at least we have a trace on the topic. I will share this with our team in order to enhance or to find a solution for that case (maybe your workaround in another way).

With regards.

closed 2024-06-07 by KRASTM