STMicroelectronics/stm32h7xx-hal-driver #32: Incorrect ADC calibration factor size

wouter-palmsens, 2023-01-23:
The macro `IS_ADC_CALFACT` is currently defined (see [here](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/12ae8efadd6d8904476b5fbd319e1c266afce734/Inc/stm32h7xx_hal_adc_ex.h#L925)) as:
```c
#define IS_ADC_CALFACT(__CALIBRATION_FACTOR__) ((__CALIBRATION_FACTOR__) <= (0x7FU))
```

This macro is used to check the parameters of `HAL_ADCEx_Calibration_SetValue` (see [here](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/12ae8efadd6d8904476b5fbd319e1c266afce734/Src/stm32h7xx_hal_adc_ex.c#L270)):

```c
  assert_param(IS_ADC_CALFACT(CalibrationFactor));
```

Consequently, when `USE_FULL_ASSERT` is enabled in the HAL, the assertion is violated when using a calibration factor with a value larger than 127.

However, the calibration factor is 11 bits, so I think the check should be `<= 0x7FFU`.

**Patch**

The corrected code should be:
```c
/**
  * @brief Calibration factor size verification (11 bits maximum).
  * @param __CALIBRATION_FACTOR__ Calibration factor value.
  * @retval SET (__CALIBRATION_FACTOR__ is within the authorized size) or RESET (__CALIBRATION_FACTOR__ is too large)
  */
#define IS_ADC_CALFACT(__CALIBRATION_FACTOR__) ((__CALIBRATION_FACTOR__) <= 0x7FFU)
```

Note that the documentation of `HAL_ADCEx_Calibration_SetValue` also mentions the 7 bits maximum instead of 11 bits maximum:
```none
  * @param CalibrationFactor Calibration factor (coded on 7 bits maximum)
```
should be:

```none
  * @param CalibrationFactor Calibration factor (coded on 11 bits maximum)
```

RJMSTM, 2023-07-26:
ST Internal Reference: 157901

TOUNSTM, 2023-12-14:
Fixed in 6e22b5b

closed 2023-12-14 by TOUNSTM