STMicroelectronics/stm32h7xx-hal-driver #37: USB HID does not work anymore after updating to latest version of stm32h7xx_hal_pcd.c

Mathullah, 2023-05-26:
Hi there,

After we updated from HAL version 1.10 to 1.11.1, USB HID no longer works in our application.

The problem could be reduced to the file `stm32h7xx_hal_pcd.c.` In the function (and further) `HAL_StatusTypeDef HAL_PCD_Start(PCD_HandleTypeDef *hpcd)` a condition has changed. The original code:

`  if ((hpcd->Init.battery_charging_enable == 1U) &&
      (hpcd->Init.phy_itface != USB_OTG_ULPI_PHY))`

was replaced by:

`if (((USBx->CID & (0x1U << 8)) == 0U) &&
      (hpcd->Init.battery_charging_enable == 1U))`

The condition is now false and causes USB to stop running. What is the reason for this change or what boundary conditions are assumed to make it work again?

Best regards
Mathias

ALABSTM, 2023-06-07:
Hello @Mathullah,

Thank you for this report. We will get back to you as soon as we analyze it further. This may take some time. Thank you for your comprehension.

With regards,

ALABSTM, 2023-07-12:
Hi @Mathullah,

This issue has been logged into an internal bug tracker at the attention of our development teams. I'll keep you informed. Thank you again for your report.

With regards,

ALABSTM, 2023-07-12:
ST Internal Reference: 157168

ALABSTM, 2024-02-19:
fixed in commit 6dd7f0e35c80b6e46ab7b77c972d2e785054d7c6

closed 2024-02-19 by ALABSTM