STMicroelectronics/stm32h7xx-hal-driver #40: The type of DMA_HandleTypeDef.Instance may be incorrect.

pic-man749, 2023-11-16:
**Describe the set-up**

- STM32H743
- CubeIDE 1.13.0

**Describe the bug**

Hi.

When I programming USART communication using DMA, I encountered a compile error when attempting to retrieve the write address of the DMA buffer. I found that the type of the `DMA_HandleTypeDef.Instance` defined in `stm32h7xx_hal_dma.h`, is set to `void*`. I suspect that the correct type would be `DMA_Channel_TypeDef*`.

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/dca6b63d9cce505f2801107f514c491d850a6cbb/Inc/stm32h7xx_hal_dma.h#L138C75-L138C75

Please look into this matter and confirm if my assumption is correct.
Thank you.

pic-man749, 2023-12-13:
Dear,

I have come to the realization that the issue I raised regarding a potential bug was, in fact, a result of an error in my own code.

I tried to use `  huart2.hdmarx->Instance->NDTR` to get write address of the DMA buffer, but it was a mistake. I made a mistake because I had used this code in the past to similar calculations on an STM32F4 board. In STM32F7,  I was able to calculate this by using the `__HAL_DMA_GET_COUNTER()` macro.

I appreciate your time and attention to the matter. Please consider this issue resolved, and feel free to close it at your earliest convenience.

Thank you for your dedication to maintaining this library, and I am grateful for the excellent work you and the team do in supporting the community.

TOUNSTM, 2023-12-15:
Hello @pic-man749,

Thank you for your contribution. Regarding the point for which you opened the issue, as it has been resolved, please allow me to close the issue.
Thank you again for your report. We are looking forward to reading from you again.

With regards,

closed 2023-12-15 by TOUNSTM