STMicroelectronics/stm32h7xx-hal-driver #42: Missing HAL_ETH_LEGACY_MODULE_ENABLED

Malcolmnixon, 2023-12-12:
**Describe the bug**
The STM32H7xx HAL release notes refer to `HAL_ETH_LEGACY_MODULE_ENABLED` in the hal conf file; however the conf template provided does not contain these entries.

Referring instead to the STM32F7xx hal conf template, we do see these symbols:
```c++
/* #define HAL_ETH_LEGACY_MODULE_ENABLED */
```

```c++
#ifdef HAL_ETH_LEGACY_MODULE_ENABLED
  #include "stm32f7xx_hal_eth_legacy.h"
#endif /* HAL_ETH_LEGACY_MODULE_ENABLED */
```

It would be beneficial to have these present in the H7 HAL.

TOUNSTM, 2023-12-13:
ST Internal Reference: 168740

TOUNSTM, 2024-01-25:
Fixed in https://github.com/STMicroelectronics/stm32h7xx_hal_driver/commit/8041954d2b7d1a57a86215e2e0f0170fd937ccbe

closed 2024-01-25 by TOUNSTM