STMicroelectronics/stm32h7xx-hal-driver #43: HAL_DMA_Abort_IT leaves handle in aborted state

drdott, 2024-01-15:
There seems to be an inconsistency in the HAL_DMA_Abort_IT() function:

- if you abort a DMA1 or DMA2 transfer then the handle gets left in the HAL_DMA_STATE_ABORT state and the abort callback is not called. The handle is not usable without reinitialising.
- if you are using a BDMA instance then the handle state gets left in HAL_DMA_STATE_READY, the abort callback is called. The handle is immediately reusable.

It almost looks like the tail end of the function has been included in the else block for the BDMA by mistake.

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/35582f63f1ab507e9d3e44e8e0db074fdb67752f/Src/stm32h7xx_hal_dma.c#L954C7-L954C7

drdott, 2024-01-22:
I think this was a misunderstanding on my part and is not a bug after all.

I see now that disabling the DMA stream as part of the HAL_DMA_Abort_IT() call causes the Transfer Complete flag to be set in the Status Register. This then causes the DMA stream interrupt and the behaviour of HAL_DMA_IRQHandler() checks for the HAL_DMA_STATE_ABORT state in the handle and performs the appropriate cleanup (call abort callback, sets handle to HAL_DMA_STATE_READY, and unlocks the handle).

Sorry for wasting your time with this.

TOUNSTM, 2024-01-22:
Hello @drdott,

Thank you for your contribution. Regarding the point for which you opened the issue, as it is resolved, please allow me to close the issue.

With regards,

closed 2024-01-22 by TOUNSTM