STMicroelectronics/stm32h7xx-hal-driver #57: ll_delayblock is available when hal qspi module is enable instead of ospi

clementfumey, 2024-05-22:
**Describe the set-up**
Seedstudio Wio Lite AI (based on STM32H725AEIX )
VSCode with  GCC

**Describe the bug**

stm32h7xx_ll_delayblock.c has the following code :

`#if defined(HAL_SD_MODULE_ENABLED) || defined(HAL_QSPI_MODULE_ENABLED)`

that made it unusable for OSPI when HAL_OSPI_MODULE_ENABLED is 1

**How To Reproduce**
Not applicable

**Additional context**
There is only 4 DLYB_TypeDef in stm32h725xx.h 2 for OSPI, 2 for SDMMC so my guess is that the correct #if defined in stm32h7xx_ll_delayblock.c should be `#if defined(HAL_SD_MODULE_ENABLED) || defined(HAL_OSPI_MODULE_ENABLED)`

KRASTM, 2024-05-27:
ST Internal Reference: 182460

KRASTM, 2024-07-23:
Fixed in commit https://github.com/STMicroelectronics/stm32h7xx_hal_driver/commit/b2b4c45001f1b6b72ae449713607896fd6ca4c6b

KRASTM, 2024-11-26:
Hello @clementfumey,

After additional testing, the fix exhibited unwanted side effects, so our team decided to revert it. However, you can make the change that best suits your application.

We apologize for the inconvenience and thank you in advance for your understanding.

With regards,

closed 2024-11-26 by KRASTM