STMicroelectronics/stm32h7xx-hal-driver #58: Release v1.11.3 breaks symbols compatibility for PTP between H and FSoC families

awojasinski, 2024-06-07:
**Describe the set-up**
 * nucleo h563zi
 * VS Code, arm-zephyt-eabi-gcc

**Describe the bug**
 * After the update to v1.11.3 symbol `HAL_ETH_PTP_CONFIGURATED` has been changed to `HAL_ETH_PTP_CONFIGURED` but only for H7 and H5. F7 and F4 still have the  `HAL_ETH_PTP_CONFIGURATED` symbol.

**How To Reproduce**
1. Indicate the global behavior of your application project
Application using PTP timestaming is zephyr are not compiling

3. The modules that you suspect to be the cause of the problem (Driver, BSP, MW ...)
BSP

4. The use case that generates the problem
Application using PTP timestaming is zephyr are not compiling

5. How we can reproduce the problem
see https://github.com/zephyrproject-rtos/zephyr/pull/73713
and compilation error https://github.com/zephyrproject-rtos/zephyr/actions/runs/9381862609/job/25833126954?pr=73713

**Additional context**
There are few options how to fix the issue:
- provide symbol translation in stm32h7xx_hal_eth.h and stm32h5xx_hal_eth.h
- revert change and bring back the symbol in mentioned file
- provide symbol translation in zephyr driver
- change symbols in HAL for F7 and F4

**Screenshots**
If applicable, add screenshots to help explain your problem.

awojasinski, 2024-06-12:
It has been fixed in zephyr with https://github.com/zephyrproject-rtos/zephyr/pull/73770

ASEHSTM, 2024-06-12:
Hello @awojasinski,

Thank you for your report, the change of this symbol has already been implemented in the F7 in HAL version v1.3.1 and will be implemented in the F4 soon. 
Additionally, Zephyr should preferably align with our latest version.

With Regards,

closed 2024-06-12 by ASEHSTM