STMicroelectronics/stm32h7xx-hal-driver #59: Device revision ID definitions

carlo-dev-git, 2024-06-10:
In file `stm32h7xx_hal.h` the device revision ID definitions are not correct for STM32H7A3xx (STM32H7_DEV_ID == 0x480UL).
According to RM0455 par.64.5.7 I tried this code:

```
#if (STM32H7_DEV_ID == 0x450UL)
#define REV_ID_Y ((uint32_t)0x1003)  /*!< STM32H7 rev.Y */
#define REV_ID_B ((uint32_t)0x2000)  /*!< STM32H7 rev.B */
#define REV_ID_X ((uint32_t)0x2001)  /*!< STM32H7 rev.X */
#define REV_ID_V ((uint32_t)0x2003)  /*!< STM32H7 rev.V */
#elif (STM32H7_DEV_ID == 0x480UL)
#define REV_ID_A ((uint32_t)0x1000)  /*!< STM32H7 rev.A */
#define REV_ID_Z ((uint32_t)0x1001)  /*!< STM32H7 rev.Z */
#define REV_ID_X ((uint32_t)0x1007)  /*!< STM32H7 rev.X */
#elif (STM32H7_DEV_ID == 0x483UL)
#warning FIXME
#endif
```

but the compilation fails in `HAL_RCCEx_CRSConfig()` where the unknown REV_ID_Y is tested also for STM32H7A3xx:
```
  if ((HAL_GetREVID() <= REV_ID_Y) && (pInit->Source == RCC_CRS_SYNC_SOURCE_USB2))
  {
    /* Use Rev.Y value of USB2 */
    value = (pInit->Prescaler | RCC_CRS_SYNC_SOURCE_PIN | pInit->Polarity);
  }
  else
  {
    value = (pInit->Prescaler | pInit->Source | pInit->Polarity);
  }
```

If I haven't missed something, it's worth checking these definitions.

TOUNSTM, 2024-06-13:
Hello @carlo-dev-git,

Thank you for your contribution. Your observation about the missing device revision ID for `0x480` is indeed correct.  An internal tracker has been logged to add these definitions. However, we will not adopt the changes you have suggested, as our aim is to maintain HAL device driver compatibility across all silicon versions: To achieve this, the `HAL_GetREVID()` API shall be used within the peripherals HAL drivers in order to verify the silicon version and act accordingly.

I hope this answer made it clearer. Thank you again for your contribution. We are looking forward to reading from you again.

With regards,

TOUNSTM, 2024-06-13:
ST Internal Reference: 184118

KRASTM, 2024-07-08:
Fixed in 21b29aa338c68333bb5e489b03a04553f8fc2c12

closed 2024-07-08 by KRASTM