STMicroelectronics/stm32h7xx-hal-driver #64: stm32h7b0 and at070tn94

amirrezabv, 2024-07-15:
**Caution**
The Issues are strictly limited for the reporting of problem encountered with the software provided in this project.
For any other problem related to the STM32 product, the performance, the hardware characteristics and boards, the tools the environment in general, please post a topic in the [ST Community/STM32 MCUs forum](https://community.st.com/s/group/0F90X000000AXsASAW/stm32-mcus) 

**Describe the set-up**
 * The board (either ST RPN reference or your custom board)
 * IDE or at least the compiler and its version

**Additional context**
If you have a first analysis or a patch proposal, thank you to share your proposal.

**Screenshots**
If applicable, add screenshots to help explain your problem.

ALABSTM, 2024-07-17:
Hi @amirrezabv,

Would you mind providing details to clarify your request please?

Thanks,

ALABSTM, 2024-07-23:
Hi @amirrezabv,

Please allow me to close this issue as no useful information shared to allow any analysis.

With regards,

closed 2024-07-23 by ALABSTM