STMicroelectronics/stm32h7xx-hal-driver #68: PTP coarse/fine update mode selection dropped from HAL_ETH_PTP_SetConfig() by mistake?

KarstenHohmeier, 2024-08-28:
The recent HAL release dropped the code for the coarse/fine update mode selection below from HAL_ETH_PTP_SetConfig().

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/7a9b4eff63551204638c15f6b9faa0e1cfea219e/Src/stm32h7xx_hal_eth.c#L1537-L1541

`ETH_MACTSCR_TSCFUPDT ` may have been confused with `ETH_MACTSCR_TSUPDT` (a normal timestamp update which happens right after) and thus be deleted. With this code missing `ETH_PTP_ConfigTypeDef.TimestampUpdateMode` will no longer be applied when configuring PTP.

ASEHSTM, 2024-09-02:
Hello @KarstenHohmeier,

Could you please clarify whether the issue with the deleted code was identified through actual problem detection (e.g., during runtime or testing) or if it was noticed through a visual inspection of the code?

With Regards,

KarstenHohmeier, 2024-09-03:
I noticed this during visual inspection when looking at the diffs after upgrading the HAL and regenerating the project.
I did a quick text search if this register gets set somewhere else after the HAL upgrade but could not find any alternate location where this was getting programmed. The corresponding read in HAL_ETH_PTP_GetConfig() was still in place though. So I suspected an accidental removal and created this issue.

ASEHSTM, 2024-09-23:
Hello,

It seems to be a mistake, and we will correct it. This section will be reverted.
ST Internal Reference: 184159

Thank you.
With Regards,

ASEHSTM, 2024-10-07:
Fixed in 98788513a15f43c9999851b6c26aa8b2fccedf7e

closed 2024-10-07 by ASEHSTM