STMicroelectronics/stm32h7xx-hal-driver #69: Defines HAL_ETH_STATE_BUSY and HAL_ETH_STATE_STARTED have the same value (0x23) breaking gState checks

KarstenHohmeier, 2024-09-08:
Looking at HAL_ETH I noticed that these defines for the HAL_ETH global state (gState) have the same value (0x23):

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Inc/stm32h7xx_hal_eth.h#L1474-L1478

This breaks these checks:

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_eth.c#L717-L721

Also the comments associated with these values are wrong/misleading. Maybe you can improve on these a little.
Here is how I understand the values after looking at how they are used in the code:

- HAL_ETH_STATE_RESET means: heth private data is not initialized at all, please call HAL_ETH_Init()
- HAL_ETH_STATE_READY means: you can register callbacks now, but to start the actual hardware and the interrupts please call HAL_ETH_Start() or HAL_ETH_Start_IT()
- HAL_ETH_STATE_STARTED means: heth is running and ready for RX and TX operations
- HAL_ETH_STATE_BUSY means: heth is currently doing something 'critical' that should not be done concurrently, because it would mess up the internal state of heth or the hardware
- HAL_ETH_STATE_ERROR means: heth encountered an error that needs external handling; see HAL_ETH_ErrorCallback() and HAL_ETH_DeInit()

ASEHSTM, 2024-09-13:
Hello @KarstenHohmeier,

Thank you for your detailed report. I have reported this issue internally.
ST Internal Reference: 191045

With Regards,

ASEHSTM, 2024-10-07:
Fixed in cf89fa21345bcf6e5d0a44e34d31d043f2ab3b7f

closed 2024-10-07 by ASEHSTM