STMicroelectronics/stm32h7xx-hal-driver #71: HAL_ETH_Transmit_IT() and HAL_ETH_Transmit() do not set HAL_ETH_STATE_BUSY before modifying TX descriptors creating race-condition

KarstenHohmeier, 2024-09-08:
`HAL_ETH_Transmit_IT()` and `HAL_ETH_Transmit()` both test gState but do not set it to HAL_ETH_STATE_BUSY before operating on the TX descriptors inside `ETH_Prepare_Tx_Descriptors()`. This creates a race-condition among themselves and among each other when used concurrently from multiple threads (e.g. preemption under FreeRTOS).

Also Test-and-Set of gState needs to be done atomically (see #70).

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_eth.c#L1002-L1009

Expected was at minimum a Test-and-Set like this:

https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_eth.c#L719-L721

ASEHSTM, 2024-09-13:
ST Internal Reference: 191088

ASAHSTM, 2025-10-28:
Hello,

Please allow me to close this issue, as the fix will be considered in the LwIP interface rather than in the HAL driver (see issue https://github.com/STMicroelectronics/stm32h7xx-hal-driver/issues/70).

Thank you for your understanding.
With Regards

closed 2025-10-28 by ASAHSTM