STMicroelectronics/stm32h7xx-hal-driver #74: FDCAN doesn't work with new release

CuboiLeo, 2024-09-09:
FDCAN doesn't work with the current release; it sends out empty messages. I found the problem in stm32h7xx_hal_fdcan.c on line 6223 and line 6235; pTxHeader ->DataLength being 0x00080000 needs to shift 16 bits to the right. After reverting this to the prior release, FDCAN can work now. 

Please check whether this is a bug or if I missed some setups.

[image]

KRASTM, 2024-09-12:
Hello @CuboiLeo,

Thank you for the report.

Just to be sure, you are using the v1.11.3 version for HAL, and hal_fdcan.c & hal_fdcan.h files?
Please, can you share more details about your project and/or config?

In the meanwhile, we will try to analyze further the issue

With regards,

KRASTM, 2024-09-13:
Hello @CuboiLeo,

I checked with our team, the change is due to a fix, because in the For loop we need the value of data length and for pTxHeader->DataLenght, we need the value in the register, so that's why we have shifted to the left (pTxHeader->DataLength << 16U).

In the last version, the **FDCAN_DLC_BYTES_x** in header file was already left shifted, and we have right shift in the function FDCAN_CopyMessageToRAM(), but now we do not need to do the shift to have the value, it's already shifted.
[image]

I did test an example with STM32H743i eval board and it works fine.

So, I suggest checking the motor driver files or your config, you can also check STM32H7xx FDCAN examples for more details.

Please, allow me to close this issue, thank you for your comprehension.

with regards,

closed 2024-09-16 by KRASTM