STMicroelectronics/stm32h7xx-hal-driver #75: HAL_ETH_ReleaseTxPacket() clears wrong bit when disabling PTP TX timestamping

KarstenHohmeier, 2024-09-09:
When cleaning up the descriptors with PTP enabled HAL_ETH_ReleaseTxPacket() tries to clear the TTSE bit in DESC2 but actually clears it in DESC3. It also uses a "magic number" to clear the bit instead of using the define ETH_DMATXNDESCRF_TTSE.

This is the wrong disable:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_eth.c#L1429-L1433

This is the corresponding enable:
https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_eth.c#L1728-L1731

So the correct disable should be:

`CLEAR_BIT(heth->Init.TxDesc[idx].DESC2, ETH_DMATXNDESCRF_TTSE);`

ASEHSTM, 2024-09-12:
Hello @KarstenHohmeier,

ST Internal Reference: 190948

ASEHSTM, 2024-10-07:
Fixed in aa891057c66d15f81ae01b2115cd96d079e6bc94

closed 2024-10-07 by ASEHSTM