STMicroelectronics/stm32h7xx-hal-driver #76: DMA HAL enables FIFO Error interrupt even if the FIFO is not enabled

multiplemonomials, 2024-09-21:
**Describe the set-up**
 * NUCLEO_H743ZI2
 * arm-none-eabi-gcc 13.2.1

**Describe the bug**
Working on the mbed-ce/mbed-os DMA SPI code for STM32H7, I observed that my code was going into the DMA error ISR handler during DMA transfers.  This happened intermittently and for no apparent reason, and the error bit was the FIFO error flag.  Eventually I found [this forum thread](https://community.st.com/t5/stm32-mcus-touchgfx-and-gui/spi-dma-error-is-occurred-when-the-other-dma-memory-to-memory-is/td-p/191590) that described the issue and the fix well.  Basically, the FIFO error flag can set under "nominal" conditions just due to the DMA controller having issues acquiring the MCU internal bus.  

While this can indicate that perhaps you need to enable the FIFO or reduce bus contention in your project, it is not an error that should fail the DMA transfer, I believe.  For my application, I simply turned off the error flag and saw perfect behavior.  (it probably helps that the SPI peripheral has its own FIFO so data won't be immediately lost of this happens

**How To Reproduce**
Unfortunately I don't have an easy way to reproduce this one, but me and that person on the forum both had it occur during "normal" application operation.  In my case, it was happening when I was doing DMA SPI transfers with a pretty standard configuration (DMA controller in direct mode).

**Additional context**
In my application, I fixed it by changing [this line](https://github.com/STMicroelectronics/stm32h7xx_hal_driver/blob/ceda3ceeca2ee77a76d2de2ee6b560ad87c90a48/Src/stm32h7xx_hal_dma_ex.c#L292) to:
```cpp
      // Only enable FIFO error if FIFO is enabled (direct mode disabled).
      // In direct mode this flag can set spuriously due to bus contention.
      if(((DMA_Stream_TypeDef *)hdma->Instance)->FCR & DMA_SxFCR_DMDIS)
      {
        ((DMA_Stream_TypeDef *)hdma->Instance)->FCR |= DMA_IT_FE;
      }
```
This way the FE flag will not generate an error interrupt if the DMA

ALABSTM, 2024-10-04:
ST Internal Reference: 192903

ALABSTM, 2024-10-04:
Hi @multiplemonomials,

Thank you for your report. The point looks relevant. It has been reported to our development teams. Let's wait for their feedback.

With regards,

TOUNSTM, 2025-05-08:
Hello @multiplemonomials,

I hope you are doing well. We have finally received feedback from our development teams. The DMA FIFO error is a known issue and should be ignored by applications when users do not use HAL drivers. The HAL PPP drivers (SPI/I2C/SAI/DCMI) are equipped to manage this DMA error, ensuring that no callback is triggered in the event of a DMA FIFO error. Therefore, it is recommended that users employ HAL drivers provided by STMicroelectronics.

The following code snippet illustrates how the HAL SPI driver handles DMA errors, specifically by ignoring FIFO errors:
```
static void SPI_DMAError(DMA_HandleTypeDef *hdma)
{
  SPI_HandleTypeDef *hspi = (SPI_HandleTypeDef *)((DMA_HandleTypeDef *)hdma)->Parent;
 
  /* if DMA error is FIFO error ignore it */
  if (HAL_DMA_GetError(hdma) != HAL_DMA_ERROR_FE)
  {
    /* Call SPI standard close procedure */
    SPI_CloseTransfer(hspi);
 
    SET_BIT(hspi->ErrorCode, HAL_SPI_ERROR_DMA);
    hspi->State = HAL_SPI_STATE_READY;
#if (USE_HAL_SPI_REGISTER_CALLBACKS == 1UL)
    hspi->ErrorCallback(hspi);
#else
    HAL_SPI_ErrorCallback(hspi);
#endif /* USE_HAL_SPI_REGISTER_CALLBACKS */
  }
}
```

### Note:
In **memory-to-memory mode**, the DMA manages requests and transfers internally, so FIFO overrun conditions are not detected. This simplifies the error handling process for such operations.

I hope this explanation provides clarity on the situation. Please allow me to close this issue.

Kind regards,

closed 2025-05-08 by TOUNSTM