STMicroelectronics/stm32h7xx-hal-driver #79: RTC bug in HAL (Clearing RTC_CR_BYPSHAD bit) - wrong retype (uint8_t)

ravo, 2024-11-11:
Dears,

I found same error as https://github.com/STMicroelectronics/stm32f4xx-hal-driver/issues/36. I forgot to mention also that these error could be in whole HAL across different MCUs. I also using STM32H7 and in 1.11.2 same error. I am not "expert" in github. Pleas apologize me if I double the original issue 36.

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/6f5e35e30ac09ba7bf44139a97b289d3df53cf9c/Src/stm32h7xx_hal_rtc_ex.c#L2712
Which produce upper 24 bits are zeroed when modifying 32 bit RTC_CR register. Please correct to
hrtc->Instance->CR &= ~RTC_CR_BYPSHAD;

Regards
Radim

ALABSTM, 2024-11-15:
Dear @ravo,

Thank you for this other notification. Actually, this has been fixed a couple of days before your post this issue, as you can see below.

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/b5fe554b9cebd2ac241fcdde18af09c0cfdf9360/Src/stm32h7xx_hal_rtc_ex.c#L2712

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/b5fe554b9cebd2ac241fcdde18af09c0cfdf9360/Src/stm32h7xx_hal_rtc_ex.c#L2744

I suggest to close this issue in this case. Thank you again for having reported.

With regards,

closed 2024-11-15 by ALABSTM

ALABSTM, 2024-11-15:
Fixed in commit 66b7103d9e146b9e8c5eeab9349855669be40d8a