STMicroelectronics/stm32h7xx-hal-driver #81: USB HCD speed definitions

carlo-dev-git, 2025-01-09:
In `usbh_conf.c` sample code the `funtion USBH_LL_GetSpeed()`
```c
USBH_SpeedTypeDef USBH_LL_GetSpeed(USBH_HandleTypeDef *phost)
{
  USBH_SpeedTypeDef speed = USBH_SPEED_FULL;

  switch (HAL_HCD_GetCurrentSpeed(phost->pData))
  {
  case 0:   //// HCD_SPEED_HIGH (ok)
    speed = USBH_SPEED_HIGH;
    break;
  case 1:   //// HCD_SPEED_FULL (duplicated)
    speed = USBH_SPEED_FULL;
    break;
  case 2:   //// HCD_SPEED_LOW (duplicated)
    speed = USBH_SPEED_LOW;
    break;
  default:
    speed = USBH_SPEED_FULL;
    break;
  }
  return  speed;
}
```
calls `HAL_HCD_GetCurrentSpeed()` from `stm32h7xx_ll_usb.c`
```c
/**
  * @brief  Return Host Core speed
  * @param  USBx  Selected device
  * @retval speed : Host speed
  *          This parameter can be one of these values:
  *            @arg HCD_SPEED_HIGH: High speed mode
  *            @arg HCD_SPEED_FULL: Full speed mode
  *            @arg HCD_SPEED_LOW: Low speed mode
  */
uint32_t USB_GetHostSpeed(USB_OTG_GlobalTypeDef const *USBx)
{
  uint32_t USBx_BASE = (uint32_t)USBx;
  __IO uint32_t hprt0 = 0U;

  hprt0 = USBx_HPRT0;
  return ((hprt0 & USB_OTG_HPRT_PSPD) >> 17);
}
```

From the documentation this function is expected to return one of the following values defined in file `stm32h7xx_hal_hcd.h`:
```c
/** @defgroup HCD_Speed HCD Speed
  * @{
  */
#define HCD_SPEED_HIGH               USBH_HS_SPEED
#define HCD_SPEED_FULL               USBH_FSLS_SPEED // duplicated ???
#define HCD_SPEED_LOW                USBH_FSLS_SPEED // duplicated ???
```
but I think this is an error.

`USB_GetHostSpeed()` should return a constant taken from `stm32h7xx_ll_usb.h`
```c
/** @defgroup USB_LL Device Speed
  * @{
  */
#define USBD_HS_SPEED                          0U
#define USBD_HSINFS_SPEED                      1U
#define USBH_HS_SPEED                          0U
#define USBD_FS_SPEED                          2U
#define USBH_FSLS_SPEED                        1U
```
and `HAL_HCD_GetCurrentSpeed()` should be documented for returning a constant taken from `stm32h7xx_hal_hcd.h` (se above).

--

From **RM0043** par.57.14.28

> Bits 18:17 **PSPD[1:0]**: Port speed
>     Indicates the speed of the device attached to this port.
>     01: Full speed
>     10: Low speed
>     11: Reserved
>     00: High speed

ALABSTM, 2025-02-06:
Hi @carlo-dev-git,

Thank you for this report. The points you mentioned have been forwarded to our development teams.

With regards,

ALABSTM, 2025-02-06:
ST Internal Reference: 202502

ALABSTM, 2025-02-06:
@carlo-dev-git,

 I just noticed that the line below is related to the device, not the host. It is confusing due to the non-straight order the definitions are listed.

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/66e5110cb6c7f7d997303db329de83edc2800dae/Inc/stm32h7xx_ll_usb.h#L380

I think that, for the host, both full and low speed are defined to be the same on **purpose**, thus the `FSLS` in the name `USBH_FSLS_SPEED`. Let's wait for the development teams' feedback.

With regards,

ALABSTM, 2025-05-07:
Hi @carlo-dev-git,

I hope you are fine. According the the development team, the description should be as shown below. We will try to have this fixed soon enough, hopefully.

```
  * @retval speed : Host port device speed
  *          This parameter can be one of these values:
  *            @arg HCD_DEVICE_SPEED_HIGH: High speed mode
  *            @arg HCD_DEVICE_SPEED_FULL: Full speed mode
  *            @arg HCD_DEVICE_SPEED_LOW: Low speed mode
```

With regards,

ALABSTM, 2025-05-20:
Fixed in commit 85d65b023f9621e417177a87553d25b7b387239f

closed 2025-05-20 by ALABSTM