STMicroelectronics/stm32h7xx-hal-driver #82: SD_SendSDStatus may cause BusFault or Overflow

elmagnificogi, 2025-03-19:
When SD init would call HAL_SD_GetCardStatus , you could see the sd_status only 16 lens.

```
HAL_StatusTypeDef HAL_SD_GetCardStatus(SD_HandleTypeDef *hsd, HAL_SD_CardStatusTypeDef *pStatus)
{
  uint32_t sd_status[16];
  uint32_t errorstate;
  HAL_StatusTypeDef status = HAL_OK;

  errorstate = SD_SendSDStatus(hsd, sd_status);
  if (errorstate != HAL_SD_ERROR_NONE)

```

In SD_SendSDStatus, when noraml read status only use 8 of 16.

```
static uint32_t SD_SendSDStatus(SD_HandleTypeDef *hsd, uint32_t *pSDstatus)
{
  SDMMC_DataInitTypeDef config;
  uint32_t errorstate;
  uint32_t tickstart = HAL_GetTick();
  uint32_t count;
  uint32_t *pData = pSDstatus;

 .....

  /* Get status data */
  while (!__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXOVERR | SDMMC_FLAG_DCRCFAIL | SDMMC_FLAG_DTIMEOUT | SDMMC_FLAG_DATAEND))
  {
    if (__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_RXFIFOHF))
    {
      for (count = 0U; count < 8U; count++)
      {
        *pData = SDMMC_ReadFIFO(hsd->Instance);
        pData++;
      }
    }

    if ((HAL_GetTick() - tickstart) >=  SDMMC_USERTIMEOUT)
    {
      return HAL_SD_ERROR_TIMEOUT;
    }
  }
```

But it will read more than the left 8 of 16，and i read the H7 reference datasheet , there is no more thing about DPSMACT, why read more FIFO DATA at the end of SD_SendSDStatus? 
```
  while ((__HAL_SD_GET_FLAG(hsd, SDMMC_FLAG_DPSMACT)))
  {
    *pData = SDMMC_ReadFIFO(hsd->Instance);
    pData++;

    if ((HAL_GetTick() - tickstart) >=  SDMMC_USERTIMEOUT)
    {
      return HAL_SD_ERROR_TIMEOUT;
    }
  }

  /* Clear all the static status flags*/
  __HAL_SD_CLEAR_FLAG(hsd, SDMMC_STATIC_DATA_FLAGS);

  return HAL_SD_ERROR_NONE;
}
```

In Fact we met a fault situation that our SD_CK not stable, which may cause DPSMACT set to 1, then when we reinit sd driver, it read the empyt FIFO(no reference says you cant read the empty FIFO ), then go to BusFault, so i check the code, found  the problem. Maybe we could check more status  avoid the fault? Could you give some advice about it ?

ASEHSTM, 2025-03-25:
Hello @elmagnificogi,

Thank you for your report.

To assist you further, could you please provide us with more information about your use case that caused the BusFault issue?

Thank you.
With Regards,

elmagnificogi, 2025-03-26:
Ok, @ASEHSTM 
It's HardFault, because we make all Fault point to HardFault, in fact it's bus fault
[image]

we make a SD reinit when sd error happended. Then it reinit , go SD_GetCardStatus then  read FIFO,make bus fault.

our SD clock wave form.It may makes the DPSMACT set 1 when reinit。This setting is not 100% likely to occur, the probability is very low, but we met 3 or 4 times
[image]

when crash，we read the sd status register
[image]

DSPMACT is 1, the RX FIFO is empty
[image]

What else do you want to know? Feel free to ask

ASEHSTM, 2025-04-03:
Hello @elmagnificogi,

The HAL_SD_GetCardStatus function uses a 16-word array to store the SD card status. In contrast, SD_SendSDStatus initially reads 8 words when the FIFO is half-full. However, the function continues to read the remaining data until the DPSM is inactive, ensuring that all required data is retrieved. This approach is recommended in the reference manual, as shown below:

       
               
[image]

Regarding the BusFault problem due to reading an empty FIFO, since the issue is related to your clock configuration, we recommend adding an additional check before each FIFO read to ensure data availability. Use the SDMMC_FLAG_RXFIFOE flag to verify data availability before reading the FIFO in both read loops. This way, the FIFO is read only when it contains valid data, preventing BusFaults.

Thank you.
With Regards,

elmagnificogi, 2025-04-03:
Thank your reply @ASEHSTM 
So HAL code will add SDMMC_FLAG_RXFIFOE flag?  We have try it before your reply, it's in test now

ASEHSTM, 2025-04-15:
Hello @elmagnificogi,

We currently have no plans to incorporate the SDMMC_FLAG_RXFIFOE flag check, as our existing implementation has been thoroughly tested and is functioning correctly. However, you may add this check to your code to address the fault caused by the unstable SD clock.

Please allow me to close this issue.

Thank you for your understanding.

With Regards,

closed 2025-04-15 by ASEHSTM