STMicroelectronics/stm32h7xx-hal-driver #83: Typo in comments of EXTI hal driver

prossin, 2025-03-26:
Just a typo in module description! ;-)

ALABSTM, 2025-04-02:
Hi @prossin,

Thank you for having reported. Would you mind precising the file and the line where this typo occurred, please?

With regards,

ALABSTM, 2025-04-04:
Hi @prossin,

Any clarification please?

With regards,

ALABSTM, 2025-04-07:
Hi @prossin,

Please allow me to close this issue as no feedback yet. Feel free to post another one with more relevant details. Thank you for your comprehension.

With regards,

prossin, 2025-04-22:
Hi @ALABSTM 
I missed your clarification request... sorry!
The file concerned are stm32h7xx_hal_exti.c (line 98, 102, 105) and stm32h7xx_hal_exti.h (line 66) where the word Exit was used instead of EXTI. That's all! :-)
Regards, Paolo.

ALABSTM, 2025-04-28:
Hi @prossin,

Thank you for your reply. You're right, some typos we missed, that I list below, based on your comment. In the **first three cases**, the word `exiting` should be replaced by `EXTI`. In the **fourth case**, it is the word `Exit` that should be replaced.

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/45c28d2982eb470621531e0d6223d5530af80fcf/Src/stm32h7xx_hal_exti.c#L98 

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/45c28d2982eb470621531e0d6223d5530af80fcf/Src/stm32h7xx_hal_exti.c#L102

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/45c28d2982eb470621531e0d6223d5530af80fcf/Src/stm32h7xx_hal_exti.c#L105

https://github.com/STMicroelectronics/stm32h7xx-hal-driver/blob/45c28d2982eb470621531e0d6223d5530af80fcf/Inc/stm32h7xx_hal_exti.h#L66

Let's reopen this issue and get these typos corrected.

With regards,

TOUNSTM, 2025-04-28:
ST Internal Reference: 208684

KRASTM, 2025-05-05:
Fixed in Commit: https://github.com/STMicroelectronics/stm32h7xx-hal-driver/commit/bd8225edaa9160b759dd3b2ebe260c5a45a3c44b

closed 2025-05-05 by KRASTM