STMicroelectronics/stm32h7xx-hal-driver #86: I couldn't use SD of 1GB probably Card V1

rod6214, 2025-05-28:
**The set-up**
 * I'm using a custom board with a STM32H743IIT6 microcontroller
 * STM32CubeIDE version 1.18.1  Build: 24813_20250409_2138

**Bug**
I just tried to read /write SD card of 1GB by using the FATS middleware but the application was returning the error HAL_SD_ERROR_UNSUPPORTED_FEATURE in the initialization.  

**How To Reproduce**
1. Open up your STM32Cube IDE and create a new STM32 project, after that, prepare SDMMC2 module to start receiving/transferring data, finally choose over your code generator .ioc, the FATS middleware for SD Card. 

2. Insert a SD Card V1 of 1GB.

3. Go to App folder inside FATS section and just use the code snippet in the function called MX_FATFS_Init, then add an additional line to mount your SD card as shown below. You will notice that f_res has FR_NOT_READY state.
`FRESULT f_res = f_mount(&fs, (TCHAR const*)SDPath, 1);`

**How to solve it**
I was able to fix this issue temporary with this [commit](https://github.com/STMicroelectronics/stm32h7xx-hal-driver/compare/master...rod6214:stm32h7xx-hal-driver:master), please let me know if this could be a good approach.

KRASTM, 2025-06-02:
Hello @rod6214,

Thank you for the report.

Could you please share the type and reference of the SD card?

With regards,

KRASTM, 2025-06-16:
Hello @rod6214,

Any updates please?

With regards,

rod6214, 2025-06-17:
Hello KRASTM,

I was able to test with different cards and it's working well for 16GB cards V2 using both SDMMC1 and SDMMC2 modules, however, there is an issue for 32GB cards due stocking on the final validation. I tested with the followwing:
16GB V2 Sandisk and Kingston. Success ✅
1GB V1 Kingston. Fail ❌
32GB V2 Sandisk and Kingston. Fail ❌

KRASTM, 2025-06-18:
Hello @rod6214,

Thank you for your feedback.

Just to be sure, for the failure cases, is it the same error within the initialization `HAL_SD_ERROR_UNSUPPORTED_FEATURE`?

With regards,

KRASTM, 2025-06-25:
ST Internal Reference: 212872

KRASTM, 2025-07-01:
Hello @rod6214,

You could submit another ticket for the SD 64GB issue, including the description.

With regards

KRASTM, 2025-10-24:
Fixed in Commit: https://github.com/STMicroelectronics/stm32h7xx-hal-driver/commit/b7935b8f12bcf931813be7ae4b8ccf822b73e58a

closed 2025-10-24 by KRASTM