STMicroelectronics/stm32h7xx-hal-driver #87: HAL_I2C_Mem_Read_DMA should call I2C_Enable_IRQ with flag I2C_XFER_RX_IT instead of I2C_XFER_TX_IT

xiankunxu, 2025-06-03:
**Describe the set-up**
 * Board: F7 nucleo board

**Describe the bug**
Bug comes when there is a `HAL_I2C_Mem_Read_DMA` followed by `HAL_I2C_Mem_Write`. 

1. When `I2C_Enable_IRQ` is called with flag `I2C_XFER_TX_IT`, after DMA read is done, TXIS is still enabled.
2. Later call of blocking write function `HAL_I2C_Mem_Write` will trigger the `I2Cx_EV_IRQHandler`, while XferISR will be NULL, so again TXIS will not get cleared, and the  `I2Cx_EV_IRQHandler` will be entered/exit infinitely. This makes other lower priority interrupts and application codes will not get executed.

**How To Reproduce**
1. Call `HAL_I2C_Mem_Read_DMA`
2. call `HAL_I2C_Mem_Write`
3. check if `I2Cx_EV_IRQHandler` is executed infinitely.

KRASTM, 2025-06-05:
Hello @xiankunxu,

Thank you for the report.

Regarding the issue, you should check whether the device is ready after the first operation `HAL_I2C_Mem_Read_DMA` and before executing another operation such as `HAL_I2C_Mem_Write`, because there is a risk of an error occurring (e.g. a Busy error), which was the result of testing your scenario.

You can refer to our example for more details. it's under STM32CubeH7\Projects\STM32H743I-EVAL\Examples\I2C\I2C_EEPROM_fast_mode_plus.

With regards,

KRASTM, 2025-06-13:
Hello @xiankunxu,

Please, allow me to close this issue, but if you have more information to share, please reopen it or post another ticket.

With regards,