STMicroelectronics/stm32h7xx-hal-driver #90: Missing MDIO Clause 45 support for STM32H7

GustavSVJ, 2025-08-04:
I am using the STM32H7 along with Zephyr and a TJA1103 phy. The phy requires Clause 45 support which the STM32H7 has but it is not included in the hal layer and therefore not available in Zephyr. 

I am willing to make the necessary code changes to get it working myself as I also have hardware available to test the changes. Before I start writing code I would like you to confirm that you are not opposed to these changes.

Kind regards,
Gustav

ALABSTM, 2025-08-06:
See also #52

GustavSVJ, 2025-08-07:
Dear Ali,

The PR you linked was rejected. Would a new PR with similar changes meet the same fate?

Kind regards,
Gustav

ASEHSTM, 2025-08-21:
please allow me to close this issue

closed 2025-08-21 by ASEHSTM